visualapproach/WiFi-remote-for-Bestway-Lay-Z-SPA #815: Lay-Z-Spa - Have they changed the connectors?!

mossc001, 2025-01-18:
I'm not sure if anyone else is able to confirm, but have Lay-Z-Spa changed their connectors in their latest manufactured models? I've got pictures of an S100102 with what appears to be slightly different connector; the pin pitch is the same. I believe it's still a JSM ST but any ideas of exact specification?

[image]
[image]
[image]

kieran-tanner-9, 2025-01-18:
Hi @mossc001 it's me from eBay (kietan21) sorry to be stalking you 😂 but I was clutching at straws trying to assist you looks like someone else has discovered this so not sure if they will be of any help 😃

https://github.com/visualapproach/WiFi-remote-for-Bestway-Lay-Z-SPA/discussions/810

mossc001, 2025-01-19:
Hi @kieran-tanner-9 , thanks!
I think they're JST-XH connectors with a 2.5mm pitch now so I'll get some ordered.

mossc001, 2025-01-27:
I can confirm they're JST-XH connectors. I bought some from Amazon and confirm a fit.

closed 2025-01-27 by mossc001

hunzamar, 2025-10-16:
Can you please post a picture about the pinout? Is it the same with both connectors? Thank you!