visualapproach/WiFi-remote-for-Bestway-Lay-Z-SPA #838: [OTHER] ESP stopped working with the display

miczu71, 2025-05-04:
**Hardware (please complete the following information):**
 - Pump model: 54154
 - 6 wires
 - PCB type: V2B
 - Level shifter type: U2

**Software (please complete the following information):**
 - Version 4.4.6

**Your message**
So I got the whole package pre-built and pre-flashed. It worked fine until yesterday, when I noticed that the screen was freezed on some value and not responsive.  I power-cycled the unit.
The story now is that the screen is off and is not reacting to any buttons when ESP is connected. I'm able to connect to the web panel and it's visible via MQTT in HA, however it shows only data readings (i.e. temperature). I can't interact with any settings.

When I unplug the ESP, the spa works fine: the display/control panel is working fine and reacting to buttons.

What I noticed, is that when I power the mains (when ESP is connected), **the display starts showing the usual data (I think it was SW release date and some other data?), but it stops in the middle of boot procedure with a click sound from the display, and then goes dark.** It doesn't respond to the power button - I can hear sound when it's pressed, but no reaction.

What I tried already:
- resetting the ESP (RST button)
- reflashing the ESP
- disconnecting from mains in various combinations

Once again - without the ESP connected, everything is fine. The device worked fine for past 2-3 months. Please help to troubleshoot :)

miczu71, 2025-05-06:
It's self-resolved after 2 days...
I packed the board into a plastic bag just in case it was moisture-related.

miczu71, 2025-05-08:
And it returned... No rain today

PZackrisson, 2025-06-01:
Did you found a solution for the problem? 

I have the exact same problem and sometimes it randomly starts work. Then after a couple of days it’s hanging and after restart is not Lightning up any more.

miczu71, 2025-06-01:
Not at all. I managed to get some help with resoldering the connectors. It was done by a professional, but the problem kept returning.

Last time I tried to make it work again (last week) it didn't fire up at all, the display was not responsive and in-app controls didn't work...

I'm still hoping this is a software-side issue.

visualapproach, 2025-06-01:
I don't think it's software issue since it worked previously. Unless you changed the config.
You can check the signal quality by clicking http://layzspa.local/debug-on/ and let it run for a while. Then click http://layzspa.local/debug-off/ and look at log.txt in http://layzspa.local/dir/

miczu71, 2025-06-05:
[

[log.txt](https://github.com/user-attachments/files/20615546/log.txt)

](url)

Here's my log. Tried to power on the spa from the display and push some buttons here and there via web panel.

visualapproach, 2025-06-06:
The CIO to/from ESP line is dead, appearantly.

clavat, 2025-06-16:
@visualapproach how do you see that?
I've the same issue, here my log file [log.txt](https://github.com/user-attachments/files/20756645/log.txt).

@miczu71 which connectors did you resoldered ?

visualapproach, 2025-09-05:
Wow, that's some corrosion!

closed 2025-11-13 by github-actions[bot]