visualapproach/WiFi-remote-for-Bestway-Lay-Z-SPA #840: [FEATURE RQ] Set Ambient Temp to temp sensor (not slider) and omit controls - if physical dallas sensor

chunkysteveo, 2025-05-09:
**Is your feature request related to a problem? Please describe.**
I have enabled the Ambient Temp sensor as a Dallas temp probe on GPIO0 on the board, which works really well to determine time to ready, virtual temp etc. But on the main controller web page, and as an MQTT entity - the ambient can be controlled to set it .up and down 

**Describe the solution you'd like**
It would be great if the "Enable Ambient Temperature" checkbox (or specifying the GPIO number) was set - that the ambient controls on the main web page were disabled/hidden, and the gauge MQTT entity was changed to an actual temperature sensor.

It's not a deal breaker, but i've accidentally flicked the slider in Home Assistant for the ambient temp sensor which then alters all the calculations, which is then adjusted back down to the "real" temperature on update - but it just looks odd that a temperature value being reported is showing as an editable one. Same with the main controller web GUI page, it just looks like you can control the ambient temperature from the GUI controls, despite it reporting the actual real temp.

I assume in the code we could check for the checkbox or GPIO number present and hide the GUI controls, and in the HA MQTT either set it as a slider, or set it as a temp sensor, if/else? Not tested that yet in my code.

More GUI niggles than anything!

visualapproach, 2025-05-16:
Good point! If nobody beats me to it I'll look into it when I get the time.

chunkysteveo, 2025-05-16:
> Good point! If nobody beats me to it I'll look into it when I get the time.

Ditto! 😂

closed 2025-07-23 by github-actions[bot]