visualapproach/WiFi-remote-for-Bestway-Lay-Z-SPA #842: Hwtest question

zupra87, 2025-05-13:
Hi, im new here. Finished building the board, run hwtest and got these errors. Is this bad soldering or what could it be?

Sending on D1, receiving on D6
---- // O errors out of 100
Sending on D2, receiving on D4
0-0-0-0-0-0-0-0-0-0-0-0-0-0-0-0-0-0-0-0-0-0-
0-0-0-0-0-0-0-0-0-0-0-0-0-0-0-0-0-0-0-0-0-0-0-
0-0-0-0-011 50 errors out of 100
Sending on D5, receiving on D3
0-0-0-0-0-0-0-0-0-0-0-0-0-0-0-0-0-0-0-0-0-0-
0-0-0-0-0-0-0-0-0-0-0-0-0-0-0-0-0-0-0-0-0-0-0-
0-0-0-0-0 11 50 errors out of 100
Sending on D6, receiving on D1
12222993223277---35
1-1-1-1- 1/ 50 errors out of 100
Sending on D4, receiving on D2
1-1-1-1-1-1-1-1-1-1-1-1-1-1-1-1-1-1-1-1-1-1-1-
1-1-1-1-1-1-1-1-1-1-1-1-1-1-1-1-1-1-1-1-1-1-1-
1-1-1-1- 1/ 50 errors out of 100
Sending on D3, receiving on D5
1-1-1-1-1-1-1-1-1-1-1-1-1-1-1-1-1-1-1-1-1-1-1-
1-1-1-1-1-1-1-1-1-1-1-1-1-1-1-1-1-1-1-1-1-1-1-
1-1-1-1- 1/ 50 errors out of 100
End of test!

visualapproach, 2025-05-13:
Welcome @zupra87 
Yes, no connection, so solders or wires or something is bad it seems.

zupra87, 2025-05-13:
I have no idea what to do next, maybe ill try to desolder all and start from beginning

visualapproach, 2025-05-13:
You can probably save it by remelting the joints. They look sub par tbh. Use flux and keep the tip on the iron clean with fresh solder on it. Then heat up the joints until the solder hugs the pins. There is several youtube tutorials on the matter.
Wish you luck

zupra87, 2025-05-14:
Thank you for the tips

zupra87, 2025-07-04:
Tried to make my solders better. Now only two errors appear that I cant seem to solve no matter what I try.

Sending on D1, receiving on D6
-
Sending on D6, receiving on
D1

50/100 error on both of these

What to try next?

closed 2025-10-07 by github-actions[bot]