visualapproach/WiFi-remote-for-Bestway-Lay-Z-SPA #857: Sometimes bad Temperature readings

fbeek, 2025-06-03:
Hello,

today I looked into my sensor graph of the spa temperature readings and found some spikes in the readings. I think its simply the case that sometimes the ESP has a problem while receiving the values from the Spa or something else which leads to an unplausible reading.

Is there any possibility to clamp these readings or build in a plausibility check? I mean -7 degrees and +60 are out of range for the spa :-D

Kind regards

[image]

DanielG557, 2025-08-16:
Hi, that would be great. I can confirm that the problem is still existing with the newer version "Firmware: 2025-07-28-1200". I also think I can hear the relay for the heating switching constantly.

craigmcfly, 2025-08-17:
> Hello, I tried to perform debugging, but I get the message “404: File not found.” I tried to open the link “layzspa.local/debug-on.” Did I do something wrong?

my fault - I missed the forward slashes off the URLs

visualapproach, 2025-08-17:
Sorry for late reply. Yes the logs show only 62% of the messages from the CIO (pump) is correct. I'm not sure if it's possible to improve by shorter wires or use of filters, or if the ESP has too much latency for the new and faster pump computers. Although some people have it work...

fbeek, 2025-09-03:
Did you mean this one?

[image]

visualapproach, 2025-09-04:
Yes, but with the temp also. 

Temp graph
Unit graph

I want to see if the spikes are correlated to unit

visualapproach, 2025-09-07:
Hi, by filters I suggested to use the B-ports with the caps and resistors. They act as low pass filters. Some need them, some don't.

closed 2025-11-14 by github-actions[bot]