visualapproach/WiFi-remote-for-Bestway-Lay-Z-SPA #874: [BUG] Lost control of spa - [abc] displayed on spa display?

KnT04d, 2025-06-27:
System has been working well for the few months since initially installed.
Noticed the app had stopped controlling the spa a couple of nights ago.
I thought I had switched it off for the two previous nights but the temp hod not dropped so was apparent spa had been running whole time over about 48 hours.
I tried to repeatedly swith the pimp and heater off within the app but could not get it to switch off.
App would show slider switches for both functions sliding to the off position and then sliding right back to the on position.
Reset the ESP and then got display [abc] on spa display.
Still able to connect wirelessly but still no control and no control on spa control panel either.
reset network setting in ESP and reconnected and reconfigured back to same wifi network successfully but still no control over spa pump or any other funstions and still no control on spa control panel.

"Saluspa Honolulu" Model #S100104.
6 Wire version
Module bought from Metaljay assembled and flashed.

[image]

[image]

metaljay, 2025-06-28:
have you checked if there is any corrosion on the pins? If it's been working fine for months and then died all of a sudden it sounds like water moisture somewhere. 

Send some photos of the board please.

KnT04d, 2025-06-29:
Have spent the afternoon troubleshooting and have come up with the following:
I can connect to the ESP "Lay-Z-Spa" wifi network directly with PC and phone
I can reset and set and save wifi settings to the module.
I cannot connect to the module vie my home wifi network.
I have tried two different wifi networks with the same results.
The blue wifi light on the module is blinking quickly and continuously.

When the module is connected to the home wifi network the "Lay-Z-Spa" wifi network still appears as an available wifi network
I have taken the module out of the spa pump and connected it to my PC via USB and it behaves the same way.
If I select the LayZSpa wifi on my PC I connect via wifi and get full communication
If I switch back to home wifi network and try to connect to mudule via web browser I get "Site can't be reached error message" etc.
I Have check in my router attached devices list and can see no sign of LayZSpa module.
I have inspected board and it looks good but I didn't make it - I was made by Metaljay and delivered complete and flashed.
Has worked well since I installed it a few months ago.
Pics attached.... 

https://github.com/user-attachments/assets/fee63fa1-54b6-4c1e-9e02-927b04d6a764
[image]
[image]

metaljay, 2025-06-29:
My suggestion is to take the board out and wash it down with isopropyl alcohol and a toothbrush. 

If you’re comfortable I’d tried re-flashing it via USB to rule out something hasn’t become corrupt too.

There doesn’t look like any corrosion anywhere, have a closer look at the joints where the cable connects to the board.

visualapproach, 2025-06-29:
Sidenotes: the blue LED has nothing to do with Wi-Fi.
The ESP can only connect to 2.4 GHz Wi-Fi.

KnT04d, 2025-06-29:
> Sidenotes: the blue LED has nothing to do with Wi-Fi. The ESP can only connect to 2.4 GHz Wi-Fi.

Yeah I know about the 2.4G - It was connected to the same wifi up until now.
I will clean and closely check the cable joints as per metaljays suggestions.

I am not sure about how to reflash the unit - where do I find instructions on this?

metaljay, 2025-06-29:
https://github.com/visualapproach/WiFi-remote-for-Bestway-Lay-Z-SPA/blob/master/bwc-manual.pdf

Worst case, ship it back to me and I’ll do it, but honestly the flashing is the easiest bit

KnT04d, 2025-07-02:
Just to check - page 12 of the manual states that if the wi-fi module is connected to a wi-fi network "If the ESP can connect to your Wi-Fi it will shut down its own "Soft" access point".
Given that the "Lay-Z-spa" wi-fi is still showing up would this suggest the issue is that the wi-fi module will not connect to a wi-fi network? Isn't this odd as it will connect to my phone or my PC through the "Soft" AP happily?
Why would it be able to talk to the wi-fi network through it's own wi-fi network but not another?
Is this a hint as to what the issue may be? possibly/probably a software issue?

I have not re-flashed the module yet. I have read through the section in the manual that covers this and I want to know if I will also need to re-install the data files as referenced in the yellow box in the manual?

metaljay, 2025-07-02:
It sounds like it’s not connecting to your home network for some reason, WPA3? Fast roaming? 2.4Ghz? Lots of things could cause this.

The test is to tether from your phone to the board, see if it connects to that.

KnT04d, 2025-07-02:
> <visualapproach/WiFi-remote-for-Bestway-Lay-Z-SPA/issues/874/3026631311@
> github.com>

Yeah it's weird as it has been connecting to my network up until recently
with no problems. It's not WPA3 or 2.4G.
And like I said I took it to work and tried to connect it to my work
network as well and no joy there either.
What do you mean by "tethering from my phone to the board" I have already
connected it multiple times to my phone directly using the LayZSpa hotspot
the board makes and it works fine.
That's why I took it to work to test it as it is a completely different
wi-fi network just in case my network at home was the problem.
It does the same thing there. Will connect via wi-fi to my phone or PC
directly but will not switch over to the normal wi-fi network and allow me
to access it via the router.

I'm no network wiz but I have over 130 devices running on my network which
has 6 Mesh devices spreading it throughout my home and a large number of
these are 2.4g which are mostly smart bulbs,switches,wall switches etc and
they all work fine.
It's the main reason my network is so over the top - so I can run all my
smart toys.
I will have a closer look at the board and connectors soon, at first glance
they all look clean as a whistle - End of financial year time right now so
I am up to my ears at work - not a lot of down time.
Thanks for the feedback
Cheers
Tony

KnT04d, 2025-07-02:
Yeah it's weird as it has been connecting to my network up until recently with no problems. It's not WPA3 or 2.4G.
And like I said I took it to work and tried to connect it to my work network as well and no joy there either.
What do you mean by "tethering from my phone to the board" I have already connected it multiple times to my phone directly using the LayZSpa hotspot the board makes and it works fine.
That's why I took it to work to test it as it is a completely different wi-fi network just in case my network at home was the problem.
It does the same thing there. Will connect via wi-fi to my phone or PC directly but will not switch over to the normal wi-fi network and allow me to access it via the router.

I'm no network wiz but I have over 130 devices running on my network which has 6 Deco Mesh devices spreading it throughout my home and a large number of these are 2.4g which are mostly smart bulbs, switches, wall switches, etc and they all work fine.
It's the main reason my network is so over the top - so I can run all my smart toys.
I will have a closer look at the board and connectors soon, at first glance they all look clean as a whistle - End of financial year time right now so I am up to my ears at work - not a lot of down time.
Thanks for the feedback
Cheers

metaljay, 2025-07-02:
I would try flashing the software again, just download visual studio code, and press 2 buttons (upload, and file system upload)

KnT04d, 2025-07-03:
Got it - was a faulty USB cable would you believe. Was providing power but no data

KnT04d, 2025-07-04:
OK - have re-flashed everything. Pretty sure it is complete and intact as all of my previous settings have defaulted back to nothing including wi-fi details and the Softspot wi-fi from the module now has a number after the lay-z-spa name. It did not before it was just "Lay-Z-Spa".
Still same issue and behaviour - can connect directly to module via wifi and enter wifi details and save/restart ESP.
Unit will not connect to local wifi and softspot "Lay-Z-Spa-9824001" wifi signal still showing as available network.
I have also tried connecting it to my phone via phone hotspot - same behaviour.
I have confirmed phone hotspot as working by connecting PC and ipad to it no probs.

I have no idea what to try next - maybe I should just buy another one? I can send this one back to @metaljay and he can see if it will work for him and then send it back to me and I'll have a spare module for next time.

[image]

metaljay, 2025-07-04:
'erase' the flash first (do this a few times), then 'upload' , then 'uploaded file system'.

The fact it's not connecting to external wifi is really odd, especially as you can connect to the board.
Can you humour me, create a dedicated 2.4GHz network from your router (disable all the fancy things like fast roaming, band steering etc etc, as basic as it comes) then try connect the wifi. 

I dont sell complete boards anymore (2nd child due in 7 weeks) but happily can send you the DIY kit if you want?

KnT04d, 2025-07-05:
Just tried that. Setup a guest network - no special options.
Exactly the same.
Can you confirm the Lay-Z-Spa-******* network should disappear once the module connects to the host network?

How much for the kit/shipping etc.
Are you cool for me to send the other one back to you to see if you can get it to work?
I will try the erase firmware route first.

KnT04d, 2025-07-05:
I also just tried setting up a basic network using WEP security instead and that was the same so it's not a security issue.