visualapproach/WiFi-remote-for-Bestway-Lay-Z-SPA #880: MQTT/Zigbee2MQTT - not communicating

MartinWatson1, 2025-07-07:
Hi, seems a daft question but just erected my tub with your egg that i used flawlessly all last year and i can access the web interface but my home assistant Zigbee2MQTT coordinator wont find the egg? 

Ive not chnaged any settings on the egg but HA has updated MQTT and Zigbee2MQTT so not sure how to trigger the egg to transmit so that Zigbee2MQTT find it and connects as you would do when connecting/re-connecting any Zigbee device. 

The HA MQTT still has the Lay-Z-spa entities as per image below. Appreciate your help. Thanks

[image]

MartinWatson1, 2025-07-07:
> can you connect via the layzspa.local? re-submit your MQTT settings from the portal

Hi, i can see the MQTT settings via a wifi connection into the egg as below and these settings have not changed since it was last connected late last year when i dismantled the tub for the winter.

[image]

metaljay, 2025-07-07:
does it say 'MQTT: CONNECTED' at the bottom of the home page? if not resubmit your password again

if still no joy, download something like MQTT explorer (im on a Mac but imagine similar tools exist) to delete the client so its forced to repush

MartinWatson1, 2025-07-07:
Thansk MetalJay for your guidance - how do i get the ESP to try and pair as i set the Z2M app in HA to permit to join but it doesnt find anything new - just wondering if the ESP is not broiadcasting?

[image]

metaljay, 2025-07-08:
This is where ChatGPT is useful. 
Sounds like you need to remind-flash it to reset everything

visualapproach, 2025-07-08:
Ok, unfortunately there is no reset wifi button.

Skeletorjus, 2025-07-16:
To clear up some confisuon, Zigbee2MQTT is a bridge between Zigbee and MQTT and has nothing to do with this, you can completely forget it in the context of this project. If HA shows the device/entities via the MQTT-integration, everything is OK.

MartinWatson1, 2025-07-16:
> To clear up some confisuon, Zigbee2MQTT is a bridge between Zigbee and MQTT and has nothing to do with this, you can completely forget it in the context of this project. If HA shows the device/entities via the MQTT-integration, everything is OK.

Thanks - in the past i have connected zigbee devices to HA via Zigbee2MQTT by pairing the devcie with my SONOF Dongle-E and it shows up there with all its attributes and entities etc. Once it appears here is then shows up in MQTT where i can start to integrate it into HA.

Skeletorjus, 2025-07-16:
> Thanks - in the past i have connected zigbee devices to HA via Zigbee2MQTT by pairing the devcie with my SONOF Dongle-E and it shows up there with all its attributes and entities etc. Once it appears here is then shows up in MQTT where i can start to integrate it into HA.

Yeah, this isn't a Zigbee device, it's WiFi and communicate directly with your MQTT bruker.