visualapproach/WiFi-remote-for-Bestway-Lay-Z-SPA #886: Havana model 54298 into Home Assistant

jon0881, 2025-07-14:
Hi, 

I have a Havana model 54298 that ive connected the device up to. I added it successfully to the wifi and the HA over MQTT. Under hardware i have CIO set to 6 wire pre2021 (air) and the same for DSP.

For the PCB i was not able to set anything to work with V1 so tried to use V2B and it seems to work better. Using HA i was able to start and stop the pump and bubbles. When it came to setting heat it made the display show different numbers. Also pressing the buttons on the unit didnt have the desired effect.

I understand on the instructions that this 54298 is ??? so what i can i troubleshoot to see if this will work?

Thanks 

Jon

metaljay, 2025-07-15:
Can you share photos of your soldering and board please 
If your using the latest board it’s V2b

metaljay, 2025-07-15:
Your cables are in the wrong order. See some photos here https://github.com/visualapproach/WiFi-remote-for-Bestway-Lay-Z-SPA/discussions/615

jon0881, 2025-07-15:
Thanks @metaljay. I just bought this from a seller on Ebay. Can you be more specific whats wrong with it so i can see if its a build issue he should fix or something specific to my model i want to use it with and should do it myslef

metaljay, 2025-07-15:
I have the same pump as you, you just need to match the photos as I linked above

mossc001, 2025-07-15:
> Your cables are in the wrong order. See some photos here [#615](https://github.com/visualapproach/WiFi-remote-for-Bestway-Lay-Z-SPA/discussions/615)

The pin outs match the connector and are correct. The colour of the cables is the only difference.

mossc001, 2025-07-15:
True, couldn't see the plugs on the end.

The 6-pin with the Red LLC works fine without.

closed 2025-11-23 by github-actions[bot]