visualapproach/WiFi-remote-for-Bestway-Lay-Z-SPA #887: ESP8266 losing MQTT IP address

MartinWatson1, 2025-07-16:
Hi, I had raised an issue #880 last week with a ESP having a WiFi issue and the guy i bought the egg from sent me a replaceemnt ESP (very kind)
However this new ESP8266 LayZspa has a seperate issue. This one stays connected to the WiFi but when i change the IP address for the 'MQTT config' and save the configuration to the ESP then navigate away from this page to another and retrun to the MQTT config page - the IP address has been removed and it stataes 'undefined'. 
Can anyone advise why the MQTT IP address is not being retained please? 
Thanks

[image]

[image]

MartinWatson1, 2025-07-16:
> The pictures are from different versions and ip are stored differently.

Yes you are correct (old image I had): however this issue remains and it scrubs the MQTT address once I’ve saved it. 

Updated photos below. 

[image]
[image]

MartinWatson1, 2025-07-17:
> What firmware version does it say on main page?

Hello, 2023-05-19-1021; Model PRE2021; RSSI: -33

MartinWatson1, 2025-07-17:
Ok will do. Thank you.

MartinWatson1, 2025-07-17:
OK i have uopdated the FW to the below. And can now adjust the MQTT IP address - just want to confirm that the IP address that i set in the 'MQTT config' page should be the same IP address as detailed on my zigbee2MQTT add on in home assistant (192.168.2.143) which differes from the Wifi IP address?

[image]

MartinWatson1, 2025-07-17:
[image]

MartinWatson1, 2025-07-17:
Yes it is - static IP of 192.168.2.143 but how do i get the ESP to pair with the SONOF coordinator and Zigbee2MQTT? On Zigbee2MQTT i start a 'permit to join' but it never sees the ESP to allow it to join the network?

visualapproach, 2025-07-17:
You don't. ESP connects to HA's MQTT server. Period.
Any co-existence with other networks are completely up to HA. The ESP is not a zigbee and it doesn't know if HA connects to sonoff or homekit bridge etc.
In case of apple homekit, HA makes it work automatically if you allow it.

visualapproach, 2025-07-17:
You're welcome

MartinWatson1, 2025-07-17:
> You're welcome

Hi, I figured it out eventually. Can you fit these ESP to any of the Bestway hot tub eggs as i have a Hawaii tub with an Model S100102 egg that i would like to fit this to? Thanks

visualapproach, 2025-07-17:
Many but not any. The ones with wifi already is not compatible.

MartinWatson1, 2025-07-19:
Thanks will have a go.