visualapproach/WiFi-remote-for-Bestway-Lay-Z-SPA #907: [BUG] Display buttons not working as expected after connecting PCB

soadzoor, 2025-08-12:
**Describe the bug**
I followed this guide to add wifi functionality to my hot tub: https://antanaitis.lt/2022/09/07/wi-fi-on-your-lay-z-spa-make-it-easy-to-remote-control

Everything went smoothly, I can even control the hot tub from the network interface, but the buttons on the pump itself no longer work as intended. The buttons are swapped randomly, and some buttons don't work at all.

Before installing the PCB:

(I made a video, but it seems I can't upload it, it fails for some reason...)

After installing the PCB:

(I made a video, but it seems I can't upload it, it fails for some reason...)

**Expected behavior**
Installing this device shouldn't break existing functionality (physical buttons on the pump)

**Hardware (please complete the following information):**
 - Pump model: S100101
 - 6 wires
 - PCB type: https://www.drot-elektro.hu/platforma-arduino/121932-wemos-d1-mini-esp8266-wifi-modul-v2.0.html
 - Level shifter type: I'm not using any level shifter

**Software (please complete the following information):**
 - Version 4.4.8

**Additional context**

[image]

visualapproach, 2025-08-12:
> Level shifter type: I'm not using any level shifter

It's not a bug until you follow specs.
You're sending 3.3V signals to the display which expects 5V

soadzoor, 2025-08-12:
> > Level shifter type: I'm not using any level shifter
> 
> It's not a bug until you follow specs. You're sending 3.3V signals to the display which expects 5V

Thank you for your quick response! I ordered the level shifter, it will probably arrive on Thursday. I'm closing this for now. If I still have problems with the level shifter I might reopen.

Thanks again, have a nice day!

visualapproach, 2025-08-12:
You have a good time too mate

soadzoor, 2025-08-14:
@visualapproach 
Hey, so I just wanted to give you an update:

I soldered 2 of these to my PCB: https://modulshop.hu/4-csatornas-szintilleszto-33-5-volt-1002
One for the D4, D3, D2 pins (CIO), the other for the D7, D6, D5, D0 pins (DSP)

I connected the 3V3 pin from the PCB to the level shifters' LV pin (each), and connected the 5V pin from the PCB to the level shifter's HV pin. I also connected the GND (level shifters) and G pin (PCB) together.

It kind of works now, but not as smoothly, and not as responsively as I expected. There are significant delays between the action and showing it on the physical display, there are still "falsy" commands being interpreted, like pressing the "bubbles" on the pump, and then it turns on just for 1 sec, then turning off, on one occasion. The next time it stays on. The temperate on the display is not really updating automatically, sometimes it can take 10-15 seconds to update (or forever), or so, it's quite random.

I'm just not sure if this state is the expected behavior if I did everything well, or if there's still something I messed up, and it should work way more reliably...?

Uploaded some photos and videos here: https://photos.app.goo.gl/361u4MMUxvUqnzPK8

visualapproach, 2025-08-15:
Okay, so it seems that the communication now works with a lot of errors instead of not at all.
If the communication with the display works you should be able to read textmessages sent to it (from spa config page-commandqueue) and pressed buttons on the display should match what is reported in the console logs (rightclick main page and inspect. Select console tab)
Likely it's the cio comms that is unreliable. Some ppl succeed and some struggle with this. Don't know why since I don't have a newer pump. It may work better with shorter cables and good solder joints. Also test with resistors and resistor/caps as filters.
No guarantee it will work though.

closed 2025-10-22 by github-actions[bot]