visualapproach/WiFi-remote-for-Bestway-Lay-Z-SPA #911: Ready in” stuck at -1d -1:00:00 despite correct ambient and target temperature

LasseTheDude, 2025-09-02:
Hi, and thanks for a great project.

I’m experiencing an issue where the “Ready in” value is always stuck at -1d -1:00:00, even though I’m providing valid values via MQTT and the module is otherwise working as expected.

What I’ve tested:

I send the following payload to layzspa/command using Home Assistant:

`{
  "CMD": 15,
  "VALUE": 14.2,
  "XTIME": 0,
  "INTERVAL": 0
`}`

This correctly sets the ambient temperature. I can see the new value reflected:

In MQTT Explorer ("AMB": 14)

In the Lay-Z-Spa Module web interface (Ambient Temp: 14)

I’ve also manually changed the target temperature (e.g. -40°C) in the web interface and made sure the heater is running.

The problem:

Despite valid values being set, the “Ready in” field still shows -1d -1:00:00 and does not update.

I’ve also tried:

Restarting the module

Re-sending the ambient temperature value

Changing values directly in the web interface

Nothing seems to make the “Ready in” value work.

Additional information:

The MQTT connection is working correctly

The module status is “Alive”

All values appear correct, except for the “Ready in” time

Earlier, I tested setting the ambient temp to -40, but have since sent valid values

HenryGelderbloem, 2025-09-03:
> Check that firmware and filesystem versions are the same. (If you only updated the firmware without also uploading the new data files you may run into problems) Also calibrate the spa on config page

Am I correct in saying the firmware version is shown at the bottom of index.html? 
I am near certain I did both (I followed the steps in the PDF), but I just want to confirm how to check the filesystem version.

craigmcfly, 2025-09-03:
> Check that firmware and filesystem versions are the same. (If you only updated the firmware without also uploading the new data files you may run into problems) Also calibrate the spa on config page

I'm seeing the same thing. I'm fairly sure I updated both, but I'm not 100%. I'll try uploading the data files again later and feed back.

LasseTheDude, 2025-09-06:
How do I upload the files, and which file should I upload? I'm not very familiar with these things. :(

visualapproach, 2025-09-06:
First check the calibration.

visualapproach, 2025-09-21:
Upload the latest data files by "build filesystem" from the pio menu in vscode.
Then either "upload filesystem" which is easier but will wipe all your settings and stats, or by uploading each file from the data/ directory by using the web ui page "upload.html"

LasseTheDude, 2025-10-23:
> Upload the latest data files by "build filesystem" from the pio menu in vscode. Then either "upload filesystem" which is easier but will wipe all your settings and stats, or by uploading each file from the data/ directory by using the web ui page "upload.html"

Can you explain this better to be? Im not so good in this stuff :( Sorry man.

closed 2026-01-12 by github-actions[bot]