visualapproach/WiFi-remote-for-Bestway-Lay-Z-SPA #917: [BUG] Error state not consistent

craigmcfly, 2025-09-21:
**Describe the bug**
Error state isn't consistent on Home Assistant.

**To Reproduce**
Trigger Error 2 on hot tub

**Expected behavior**
Error code should stay at 2

**Log**

[image]

**Hardware (please complete the following information):**
 - Miami 2021
 - 6 wires
 - PCBv2
 - don't know what a level shifter type is 

**Software (please complete the following information):**
 - Firmware: 2025-07-28-1200

craigmcfly, 2025-09-30:
In case it helps, on the native controller it just flashes E02. So it's possible the period where the LEDs are off is being misinterpreted by your software?

visualapproach, 2025-10-07:
Couldn't reproduce but added a check for invalid characters when error is displayed. You are welcome to try
FW_VERSION "2025-10-07-2100" (dev branch)